MicrosoftDocs/windows-driver-docs-ddi #315: Last parameter type mistake

kl-andrey-redko, 2018-12-05:
There is a mistake about last parameter CallbackRegistration: type should be PVOID* instead of PVOID.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1a8a7db6-d3fb-928c-67ed-44872b418216
* Version Independent ID: b2d6c707-dae9-df69-dd12-e9c1918be5b1
* Content: [IoRegisterContainerNotification function - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-ioregistercontainernotification)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-ioregistercontainernotification.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-ioregistercontainernotification.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2018-12-05:
Thanks @kl-andrey-redko.
@tedhudek, not sure why we're not picking up that this is a pointer. Definition shows it is, to my untrained eye:

NTSTATUS
IoRegisterContainerNotification(
    \_In_ IO_CONTAINER_NOTIFICATION_CLASS NotificationClass,
    \_In_ PIO_CONTAINER_NOTIFICATION_FUNCTION CallbackFunction,
    \_In_reads_bytes_opt_(NotificationInformationLength) PVOID NotificationInformation,
    \_In_ ULONG NotificationInformationLength,
    **\_Out_ PVOID CallbackRegistration**
    );

tedhudek, 2019-06-05:
Hi @kl-andrey-redko sorry for the delay.  The syntax block on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-ioregistercontainernotification looks the same as the syntax block in wdm.h, shown above.  Please reopen if I'm missing something.

closed 2019-06-05 by tedhudek