MicrosoftDocs/windows-driver-docs-ddi #318: Undocumented behaviour

maharmstone, 2018-12-09:
LXSS can pass -2 (fffffffffffffffe) for the times. Can we please have it documented exactly what this means?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 30d6a301-2028-bba5-0e43-092824810891
* Version Independent ID: 5c8c373a-b34a-5e73-8002-57076540e6ce
* Content: [_FILE_BASIC_INFORMATION - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_file_basic_information)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_file_basic_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_file_basic_information.md)
* Product: **windows-hardware**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

rodwiddowson, 2018-12-09:
See also [this link](url
https://community.osr.com/discussion/comment/290374)

I have scoured the header for appropriation conte t and found nothing so there have to be code changes as well.

maharmstone, 2018-12-09:
Cheers Rod. I can see that NtfsSetBasicInfo is clearing a flag, so that thread is probably right.

rodwiddowson, 2018-12-10:
FWIW I would trust an analysis by @ladislav-zezula

Notwithstanding this behavior should be documented and the symbolic values made available in the WDK

tedhudek, 2019-06-12:
Apologies for the delay on this one.  I've been unable to get more info on the -2.  Recommend you try http://www.osronline.com/showlists.cfm?list=ntdev if it's still an issue.  If you figure it out, please use the Edit button on a docs page to open a PR :)

rodwiddowson, 2019-06-12:
@tedhudek FWIW: The initial report came from ntfsd !(a much better resource for file systems that NTDEV).
https://community.osr.com/discussion/comment/290374  That describes what it does (which is effectively to unto the -1 thing).  I'd rather not add the text since I don't have code access and I'd be making it up based on empirical evidence.

tedhudek, 2019-06-17:
We got an answer!  Assigning to Lori to add it

lorihollasch, 2019-06-20:
@maharmstone , thanks for pointing out that meaning of -2 was missing! I've updated the page to answer the mystery. Tagging @rodwiddowson since you're interested. :)

closed 2019-06-20 by lorihollasch

maharmstone, 2019-06-21:
Thanks for sorting this!