MicrosoftDocs/windows-driver-docs-ddi #327: _UNICODE_STRING.Buffer member documentation incomplete

cxw42, 2018-12-28:
In [_UNICODE_STRING - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wudfwdm/ns-wudfwdm-_unicode_string), `Buffer` can be a `USHORT*` or ` PWCH`.

- The conditions on the `#if` deciding which it is are not listed or documented.  Would you please add documentation, or a reference to how to find out (other than UTSL :) )?
- The two `Buffer` options don't have any details, and are listed as `Buffer.#4025480185` and `Buffer.#407591603`.  Would you please clarify the labels and explain the difference?

Thank you for considering this request!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ba18d032-ca0a-f967-c461-f2715a39bfa7
* Version Independent ID: da0c0414-8b5c-4249-b3a7-2400cd53804f
* Content: [_UNICODE_STRING - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wudfwdm/ns-wudfwdm-_unicode_string)
* Content Source: [wdk-ddi-src/content/wudfwdm/ns-wudfwdm-_unicode_string.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wudfwdm/ns-wudfwdm-_unicode_string.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

tedhudek, 2019-06-05:
Hi again @cxw42, funny you should ask.  This page has been the subject of some discussion lately.  We have a known rendering bug in the ref doc pipeline that results in the hash values you noticed.

After you opened the issue, the page moved to https://docs.microsoft.com/en-us/windows/desktop/api/ntdef/ns-ntdef-_unicode_string.  That reminds me, I need to put a redirect in place for that.

I do not know what the MIDL_PASS qualifier does, but would recommend that you ask on http://www.osronline.com/showlists.cfm?list=ntdev if this is still an issue for you.

For our reference, here is the definition from ntdef.h (which actually ships in the SDK, not the WDK):

```
typedef struct _UNICODE_STRING {
    USHORT Length;
    USHORT MaximumLength;
#ifdef MIDL_PASS
    [size_is(MaximumLength / 2), length_is((Length) / 2) ] USHORT * Buffer;
#else // MIDL_PASS
    _Field_size_bytes_part_opt_(MaximumLength, Length) PWCH   Buffer;
#endif // MIDL_PASS
} UNICODE_STRING;
```

closed 2019-06-05 by tedhudek

tedhudek, 2019-06-05:
Update -- the redirect from the old URL https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wudfwdm/ns-wudfwdm-_unicode_string is now in place

cxw42, 2019-06-05:
@tedhudek Thanks for handling this and the other two issues!  :up:

tedhudek, 2019-06-05:
Glad to help :)

Thanks for the sharp eye!