MicrosoftDocs/windows-driver-docs-ddi #330: argumet types not specified

gsudhakar1234, 2018-12-29:
The prototype is written like macro declaration in source code rather than for documentation. Please correct it and specify the argument type also as part of the prototype. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 831fb347-ca21-4bca-a0c1-a1dd4b0155ec
* Version Independent ID: b1ecd8b4-dfc0-7564-b6ec-88407ff1ad37
* Content: [KeAcquireSpinLock macro - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-keacquirespinlock)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-keacquirespinlock.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-keacquirespinlock.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

gsudhakar1234, 2018-12-29:
Also parameter names should be meaningful rather than a and b.

tedhudek, 2019-06-10:
Hi @gsudhakar1234 thanks for spotting this.  I've got the arg names in now and just pushed a fix that I think will add the types, should be live within the hour

closed 2019-06-10 by tedhudek