MicrosoftDocs/windows-driver-docs-ddi #339: IRQL

kouzhudong, 2019-01-08:
我的一个同事看到：&lt; DISPATCH_LEVEL
认为此函数不能运行在DISPATCH_LEVEL。
我认为只要是非分页内存，可以运行在任意IRQL上。

如果我说的正确，请修改：
IRQL	"&lt; DISPATCH_LEVEL (see Remarks section)"
为
IRQL	"&lt;= DISPATCH_LEVEL (see Remarks section)"
或者
Any level

https://correy.webs.com

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e8a2c507-a2e6-8907-d3e6-ace0f2594d7c
* Version Independent ID: c35a6618-53db-ca99-cb76-aa2ec5d905f8
* Content: [RtlLookupElementGenericTable function - Windows driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/nf-ntddk-rtllookupelementgenerictable)
* Content Source: [wdk-ddi-src/content/ntddk/nf-ntddk-rtllookupelementgenerictable.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/nf-ntddk-rtllookupelementgenerictable.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2019-03-06:
Thank you, @kouzhudong, and I apologize for letting this Issue sit for so long.
@lorihollasch, can you look into the correct dispatch level , as requested?

One of my colleagues saw: < Dispatch_level
Think this function cannot run in Dispatch_level.
I think as long as it's non-paging memory, it can run on any IRQL.
If I am right, please modify:
IRQL  "< dispatch_level (see Remarks section) "
For
IRQL  "<= dispatch_level (see Remarks section) "
Or Any level

lorihollasch, 2019-04-25:
@kouzhudong, thanks for reporting. I've updated the IRQL section to point to Remarks, where there is clarifying information about IRQL for paged versus nonpaged memory/code.

closed 2019-04-25 by lorihollasch