MicrosoftDocs/windows-driver-docs-ddi #345: DEBUG_ADDSYNTHMOD_ZEROBASE not described

kITerE, 2019-01-11:
Along with DEBUG_ADDSYNTHMOD_DEFAULT, modern DbgEng SDK contains DEBUG_ADDSYNTHMOD_ZEROBASE.
DEBUG_ADDSYNTHMOD_ZEROBASE flag, unfortunately, is not described.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6aa2a46d-884e-a7a5-a078-5376edc76854
* Version Independent ID: b519caa4-eb83-4b87-fcfb-5b3bf5229b93
* Content: [IDebugSymbols3::AddSyntheticModule - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgeng/nf-dbgeng-idebugsymbols3-addsyntheticmodule)
* Content Source: [wdk-ddi-src/content/dbgeng/nf-dbgeng-idebugsymbols3-addsyntheticmodule.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgeng/nf-dbgeng-idebugsymbols3-addsyntheticmodule.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

kITerE, 2019-01-11:
dbghelp.h:
```
//
// flags found in SYMBOL_INFO.Flags
//
<…>
#define SYMFLAG_SYNTHETIC_ZEROBASE  0x00200000
```

For **SYMFLAG_SYNTHETIC_ZEROBASE**, too, there is no description (https://docs.microsoft.com/windows/desktop/api/dbghelp/ns-dbghelp-_symbol_info)

DOMARS, 2019-06-11:
@kITerE - Working with the dev team to add updated information. Thanks for pointing out that these topics need to be updated.

DOMARS, 2020-10-29:
@kITerE - Sorry for the delay here. The dev team indicated that this flag allows to override the default behavior in some one off internal situations. So we won't be documenting this flag.  As another option when you encounter issues like this, you can ask about what you are seeing on the OSR debugger forum.

•	[OSR's windbg List](https://community.osr.com/categories/windbg)

Good luck with your work in the debugger.

closed 2020-10-29 by DOMARS