MicrosoftDocs/windows-driver-docs-ddi #346: IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX - incorrect advice?

tonysepia, 2019-01-12:
The article on IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX  suggests:
"IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX is a user-mode I/O control request. This request targets the USB hub device (GUID_DEVINTERFACE_USB_HUB). Do not send this request to the root hub."

However, the official documentation to USB Tree View utility (https://github.com/Microsoft/Windows-driver-samples/tree/master/usb/usbview) states that:
"Enumerate Downstream Ports. Given a handle to an open hub and the number of downstream ports on the hub, send the hub an IOCTL_USB_GET_NODE_CONNECTION_INFORMATION request for each downstream port of the hub to get info about the device (if any) attached to each port. "

Should we remove the "Do not send this request to the root hub" from documentation?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 210d1acd-67a1-6a3d-adeb-84bf6048c483
* Version Independent ID: 4a65abb1-9f7a-c852-404e-1a76ce9874cc
* Content: [IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_information_ex)
* Content Source: [wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_information_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_information_ex.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-01-14:
Hi @tonysepia, thanks for the question. I believe the confusion here might stem from the distinction between the root hub and other USB hubs. 

In the ReadMe of the USB tree view sample, Step 2 says "Enumerate Hubs (Root Hubs and External Hubs). Given the name of a hub, use CreateFile() to open the hub. Send the hub an *IOCTL\_USB\_GET\_NODE\_INFORMATION* request to get info about the hub...". Meanwhile, Step 3, says "Enumerate Downstream Ports. Given a handle to an open hub and the number of downstream ports on the hub, send the hub an *IOCTL\_USB\_GET\_NODE\_CONNECTION\_INFORMATION* request for each downstream port of the hub..." (emphasis added)

So, the doc seems to be correct. Does this help?

tonysepia, 2019-01-14:
I may be wrong, but the code actually sends IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX to the root hub, doesn't it?

There's also another problem in the sample documentation, so I think it may be a bit obsolete: https://github.com/Microsoft/Windows-driver-samples/issues/333

tonysepia, 2019-01-14:
This is part of the code where the example gets the root hub for the Host controller:
https://github.com/Microsoft/Windows-driver-samples/blob/6c1981b8504329521343ad00f32daa847fa6083a/usb/usbview/enum.c#L570

And the code above then calls EnumerateHub, which uses IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX

Again, unless I am mistaken - I am new to this subject!

duncanmacmichael, 2019-01-15:
No worries, I'm new to USB, too (I usually own the networking docs). :) As far as I can see, `EnumerateHub()` does not use IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX. In this order, it uses these IOCTLs in `DeviceIoControl` after calling `CreateFile()`:

1. IOCTL_USB_GET_NODE_INFORMATION
2. IOCTL_USB_GET_HUB_INFORMATION_EX
3. IOCTL_USB_GET_HUB_CAPABILITIES_EX

So IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX is still not sent to the root hub. All instances of that IOCTL are within the call to `EnumerateHubPorts()`, which does not target the root hub.

tonysepia, 2019-01-15:
Thank you very much for exaplaining! It makes sense now!

Now I am not sure whether my other comment is also a result of misunderstanding: https://github.com/Microsoft/Windows-driver-samples/issues/333 😄

duncanmacmichael, 2019-01-15:
Happy to help! In the latter case, I'm not sure. I do see them using `SetupDiEnumDeviceInterfaces()` in the sample code with the comment "Iterate over host controllers using the new GUID based interface," which is probably what led to your question. I'd keep that one open so we can see if someone from the USB team could answer it.

mmgu, 2019-01-22:
> 
> So IOCTL_USB_GET_NODE_CONNECTION_INFORMATION_EX is still not sent to the root hub. All instances of that IOCTL are within the call to `EnumerateHubPorts()`, which does not target the root hub.

Are you sure? I read the code again and EnumerateHubPorts() does target the the root hub. (The handle hHubDevice passed to it from EnumerateHub() is returned from CreateFile() on the root hub name)

mmgu, 2019-01-23:
@tonysepia You may missed my point. What I mean is that according to the code, the statement "Do not send this request to the root hub" in documentation should be removed.

tonysepia, 2019-01-23:
@mmgu Apologies! I have misread your response. I am new to C++ so have blindly trusted someone with more experience. Let's reopen this and get some clarity

duncanmacmichael, 2019-01-24:
Upon further tracing of the code, it looks like you're right. Let me see what I can find out; I'm not sure why that statement would be on this IOCTL doc page, either.

duncanmacmichael, 2019-01-24:
Hi @tonysepia and @mmgu, thanks for waiting. I contacted some of the USB devs and got confirmation that this IOCTL can, in fact, be sent to the root hub with no issues. Sorry for the confusion. I'll go ahead and remove that sentence from the docs, and that change should be live later today.

Thanks for helping to improve the docs!

closed 2019-01-24 by duncanmacmichael