MicrosoftDocs/windows-driver-docs-ddi #352: no such function ZwSetInformationFileEx

binarycrusader, 2019-01-16:
There is no such function named ZwSetInformationFileEx; it's actually named ZwSetInformationFile.  There is no Ex version of that function.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dff35e66-2d48-4668-00cd-2e0db3d12618
* Version Independent ID: 1637c278-b051-28e0-8c22-3256777e9218
* Content: [_FILE_DISPOSITION_INFORMATION_EX - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_file_disposition_information_ex)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_file_disposition_information_ex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_file_disposition_information_ex.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

metathinker, 2019-03-21:
@binarycrusader - please check again as Ted Hudek and I recently made a change that should fix the wrong reference: 47ef23878f03e04f08a13584e8b2d2f09a955deb

tedhudek, 2019-03-21:
The changes are live now.  Thanks @binarycrusader for pointing this out, and @metathinker for all the help drafting the update

closed 2019-03-21 by tedhudek

binarycrusader, 2019-03-22:
Looks good, although now the ZwSetFileInformation page needs to mention FileDispositionInformationEx.  I'll open a separate bug for that.