MicrosoftDocs/windows-driver-docs-ddi #353: Wrongly written characters

GreatFireWall, 2019-01-18:
[FwpsInjectTransportSendAsync1](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fwpsk/nf-fwpsk-fwpsinjecttransportsendasync1)

[image]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 32e58499-3281-f443-761a-a171b1aa376e
* Version Independent ID: ea2ab0ec-43f0-f67e-9b38-57c35d58c94a
* Content: [FwpsInjectTransportSendAsync1 function - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fwpsk/nf-fwpsk-fwpsinjecttransportsendasync1)
* Content Source: [wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjecttransportsendasync1.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fwpsk/nf-fwpsk-fwpsinjecttransportsendasync1.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-01-18:
Good catch @GreatFireWall! Looks like this was a goof that happened during a content migration process we did last year. I'll go ahead and fix this now, so that change should be live soon.

Also, if you'd like, for simple fixes like this feel free to use the "Edit" button on the doc, make the change, and submit a pull request. :)

Thanks!

closed 2019-01-18 by duncanmacmichael

GreatFireWall, 2019-01-19:
👌