MicrosoftDocs/windows-driver-docs-ddi #355: void Arg1 seems incorrect

gsudhakar1234, 2019-01-23:
Pl check the parameters

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6aec7611-66ac-b728-9147-366b1a99e32c
* Version Independent ID: a69f4194-cc00-cbb6-339d-bf28379d5908
* Content: [PDEBUG_EXTENSION_UNINITIALIZE - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgeng/nc-dbgeng-pdebug_extension_uninitialize)
* Content Source: [wdk-ddi-src/content/dbgeng/nc-dbgeng-pdebug_extension_uninitialize.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgeng/nc-dbgeng-pdebug_extension_uninitialize.md)
* Product: **windows-hardware**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

BMorgan1296, 2019-01-23:
Given code doesn't work:

error C2182: 'Arg1': illegal use of type 'void'
error C2365: 'PdebugExtensionUninitialize': redefinition; previous definition was 'data variable'
note: see declaration of 'PdebugExtensionUninitialize'

This is from copying the code straight in and replacing the ". . ." with some code - if the usage is different please do correct me.

DOMARS, 2019-01-26:
@gsudhakar1234 - This seems to be an issue with the new system we are using to generate the signature. This callback function has no parameters and does not return a value.

DOMARS, 2020-10-28:
Sorry for the length delay @gsudhakar1234. The topic now indicates that this his callback function has no parameters and does not return a value. And I believe the automatic generated signature is correct. For signatures, you can also examine the dbgeng.h header.

If you have additional questions, you may have success posting in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
•	[OSR's ntdev List](https://community.osr.com/categories/ntdev)
•	[OSR's windbg List](https://community.osr.com/categories/windbg)

Thanks again for reaching out and good luck with your work.

closed 2020-10-28 by DOMARS