MicrosoftDocs/windows-driver-docs-ddi #358: A void macro would not return any values.

alfaunits, 2019-01-29:
The macro should return a BOOLEAN/BOOL, it is not returning a void value.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6260b0d3-985d-5996-37cc-c6bd6d56df1a
* Version Independent ID: fc393953-7eb7-d406-8de2-0fdf77a0781c
* Content: [RtlCheckBit macro (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlcheckbit)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlcheckbit.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlcheckbit.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

alfaunits, 2019-01-29:
More of a bump, considering the last report is 3 months old.

tedhudek, 2019-06-10:
Thanks @alfaunits this is fixed now

closed 2019-06-10 by tedhudek