MicrosoftDocs/windows-driver-docs-ddi #359: Mangled content

alfaunits, 2019-01-30:
The article starts with unformatted, and probably unneeded data.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0fe2f032-89db-c787-ee72-ecca90fb1fe0
* Version Independent ID: fd400891-666b-3937-e51e-dfd1d1db9d25
* Content: [req.typenames: FILE_FS_SECTOR_SIZE_INFORMATION, *PFILE_FS_SECTOR_SIZE_INFORMATION - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_file_fs_sector_size_information)
* Content Source: [wdk-ddi-src/content/ntddk/ns-ntddk-_file_fs_sector_size_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntddk/ns-ntddk-_file_fs_sector_size_information.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

EliotSeattle, 2019-01-30:
Thanks, @alfaunits 
Looks like a rendering bug in all structs in this header. Investigating.

tedhudek, 2019-01-30:
Thanks @alfaunits this should be fixed now.  Please let us know if you see any exceptions

closed 2019-01-30 by tedhudek

1-0-9-1, 2019-01-30:
Thanks I fond out just how fast the can be. Could you let the dean know khow i it 

How it looks with the swich of users and running around but I whated to long to run my erans and I had to chang users to get to other browser that didn’t  have what I needid