MicrosoftDocs/windows-driver-docs-ddi #361: Missing description for HidNotifyPresence

reuvenab, 2019-01-31:
Page lacks  information about HidNotifyPresence function 
which is declared in HidPort.h

```
#if(NTDDI_VERSION>=NTDDI_WINXPSP1) // Available on XP SP1 and above
NTSTATUS
HidNotifyPresence(
    _In_ PDEVICE_OBJECT DeviceObject,
    _In_ BOOLEAN IsPresent
    );
#endif
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9861135a-f7ea-5b79-94bc-2bd5da065a7d
* Version Independent ID: 395b68be-b887-1631-2141-2f39fd1f2418
* Content: [hidport.h header - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/hidport/)
* Content Source: [wdk-ddi-src/content/hidport/index.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/hidport/index.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

reuvenab, 2019-01-31:
Additional issue that not directly related to the documentation, rather than code by itself:

function doesn't declared with "C" linkage with **extern** keyword
However it appears in the library file hidclass.lib as "C" linkage
which makes it impossible to use it in C++ projects.

EliotSeattle, 2019-01-31:
Thank you so much, @reuvenab.
Working on getting this in our build.
For the issues about the code, your best bet is to use the button at the bottom of the page that says, "Product feedback." I can take care of the docs, but that tool will route your concerns to the PM/Devs writing the code.

EliotSeattle, 2019-02-04:
And after researching this API, apparently that function is intended for internal use only. I will create a topic that says so and have it link from the hidport.h table of functions.

Thanks again,
Eliot

EliotSeattle, 2019-02-06:
The topic is live. Thanks for pointing this out, @reuvenab, and for raising the issue in the code.

closed 2019-02-06 by EliotSeattle