MicrosoftDocs/windows-driver-docs-ddi #366: Graphics WDK docs seems to be corrupted / missing

g0blue, 2019-02-05:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3dukmdt/ns-d3dukmdt-_d3dddi_synchronizationobjectinfo2

Most of the actual documentation for this structure is missing. Other D3DKMT / D3DDDI structures have similar issues.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a7a8a85a-a25b-7976-1943-a05fbbf5c3f3
* Version Independent ID: 0f41c087-22dd-8321-8c58-66bc9aa0200e
* Content: [_D3DDDI_SYNCHRONIZATIONOBJECTINFO2 (d3dukmdt.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3dukmdt/ns-d3dukmdt-_d3dddi_synchronizationobjectinfo2)
* Content Source: [wdk-ddi-src/content/d3dukmdt/ns-d3dukmdt-_d3dddi_synchronizationobjectinfo2.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3dukmdt/ns-d3dukmdt-_d3dddi_synchronizationobjectinfo2.md)
* Product: **windows-hardware**
* GitHub Login: @mimisasouvanh
* Microsoft Alias: **misaso**

mimisasouvanh, 2019-02-05:
@g0blue Hi Dan, We are aware of this issue and currently have a bug open to fix it. In order to see the actual content, please click Edit on the upper right corner of the topic.

closed 2019-02-05 by mimisasouvanh

duncanmacmichael, 2019-02-05:
This is now fixed.