MicrosoftDocs/windows-driver-docs-ddi #368: Missing information

paulsinnett, 2019-02-06:
This documentation doesn't say what the required buffer size is or what any of information returned would represent. If I call it on an Xbox One, Xbox360 or Playstation 4 controller it always returns false.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b7daa60f-22c8-eecb-0d00-dba12319835c
* Version Independent ID: 0b366990-6367-03ce-8228-ed0745679fdb
* Content: [HidD_GetPhysicalDescriptor function (hidsdi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/hidsdi/nf-hidsdi-hidd_getphysicaldescriptor)
* Content Source: [wdk-ddi-src/content/hidsdi/nf-hidsdi-hidd_getphysicaldescriptor.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/hidsdi/nf-hidsdi-hidd_getphysicaldescriptor.md)
* Product: **windows-hardware**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

duncanmacmichael, 2019-03-13:
Assigning to Eliot since the wdkbot seems to have missed this one.

billlattimer, 2021-02-03:
@paulsinnett finally closing this out for you and updating our Docs.

- The required buffer-size is not exposed via any API.  The only way to call the API is to give it a big-buffer (say 1KB) and increase it if the return value indicates a too-small buffer.
- The data returned in the buffer is a raw PhysicalDescriptor as described in (https://www.usb.org/sites/default/files/hid1_11.pdf#page=53).  We don’t provide any  APIs for parsing this, hence giving access to the raw buffer.
- HidD_GetPhysicalDescriptor will query the device for the HID PhysicalDescriptor (see https://www.usb.org/sites/default/files/hid1_11.pdf#page=53), not a HID ReportDescriptor (https://www.usb.org/sites/default/files/hid1_11.pdf#page=24).  When we last examined the Xbox/Playstation controllers, they did not have PhysicalDescriptors, only ReportDescriptors.  ReportDescriptors are what most developers are interested-in as it contains button/axis/haptics, but there is no API to access the raw bytes of this descriptor.  Physical Descriptors prescribe which body-part/s should use a control, and they are very uncommon.  This is exposed as we don’t give APIs for parsing this descriptor.

closed 2021-02-03 by billlattimer