MicrosoftDocs/windows-driver-docs-ddi #398: Lost a lot of documentation

DragonGeo2, 2019-02-19:
A large number of documentation pages (at least 83) relating to this topic (the shader format for shader models 1, 2, and 3) have recently gone missing with the newest MSDN move. For instance, this page used to include a hyperlink to all of the instructions which would describe in much greater detail how the D3DSIO_LIT instruction expects the source parameter token passed to it to be set up in a particular way:
src.x = N dot L    ; The dot product between normal and direction to light (for diffuse lighting)
src.y = N dot H    ; The dot product between normal and half vector (for specular lighting)
src.z = ignored    ; This value is ignored
src.w = exponent   ; The value must be between -128.0 and 128.0 (for specular lighting)

As another example, there's now no documentation describing how the D3DSIO_DST function behaves with respect to its source and destination parameter tokens (which have a special layout, somewhat similarly to the D3DSIO_LIT instruction).
Also the sample pseudocode/code for each of these instructions is now missing as well.

Two of these now-missing pages used to exist at the now-deprecated old-style URL's which redirect to this page:
https://msdn.microsoft.com/en-us/library/windows/hardware/ff568199(v=vs.85).aspx (for the LIT instruction)
https://msdn.microsoft.com/en-us/library/windows/hardware/ff557317(v=vs.85).aspx (for the DST instruction)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c0b314bf-4527-3c2d-8f8c-1fe614d623bc
* Version Independent ID: a65a46cd-422b-25b6-d7a7-f21fb2929515
* Content: [_D3DSHADER_INSTRUCTION_OPCODE_TYPE (d3d9types.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/d3d9types/ne-d3d9types-_d3dshader_instruction_opcode_type)
* Content Source: [wdk-ddi-src/content/d3d9types/ne-d3d9types-_d3dshader_instruction_opcode_type.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/d3d9types/ne-d3d9types-_d3dshader_instruction_opcode_type.md)
* Product: **windows-hardware**
* GitHub Login: @mimisasouvanh
* Microsoft Alias: **misaso**

mimisasouvanh, 2019-02-19:
@DragonGeo2 Thank you for the issue. I am currently investigating the missing information.

DragonGeo2, 2019-02-19:
Oh, I think that I've found the missing documentation. It's now located here and here:
https://docs.microsoft.com/en-us/windows/desktop/direct3dhlsl/dx9-graphics-reference-asm-vs-instructions
https://docs.microsoft.com/en-us/windows/desktop/direct3dhlsl/dx9-graphics-reference-asm-ps-instructions
The hyperlinks from each instruction opcode to their corresponding function pages may still be broken, but at least the documentation isn't lost. If anything, it may have simply been not indexed by search engines properly.

mimisasouvanh, 2019-02-20:
I'm glad to hear that. I will continue to research missing info on the opcode's and add links wherever necessary. I'll close the issue once I'm done.

closed 2019-02-22 by mimisasouvanh