MicrosoftDocs/windows-driver-docs-ddi #401: Link to MONITOR incorrect

JonAxtell, 2019-02-27:
Link to the MONITOR structure is incorrect, it does not go to the print monitor structure.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 831bb018-95bd-ec7e-9045-e13471f984fd
* Version Independent ID: c1ee603e-54af-046a-b343-86fb89d040c3
* Content: [_MONITOREX (winsplp.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/winsplp/ns-winsplp-_monitorex)
* Content Source: [wdk-ddi-src/content/winsplp/ns-winsplp-_monitorex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/winsplp/ns-winsplp-_monitorex.md)
* Product: **windows-hardware**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2019-08-01:
Link updated.

barrygolden, 2019-08-01:
Thanks @JonAxtell for opening an issue on the broken link. Link has been fixed.

closed 2019-08-01 by barrygolden