MicrosoftDocs/windows-driver-docs-ddi #404: CreatePipe and CreateMailslot parameters not described

BrianCatlin, 2019-02-28:
Where is the description for CreatePipe and CreateMailslot?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 90b8d975-b4f8-c43f-f91f-fad1bb5a5cee
* Version Independent ID: 8efd9348-dcb5-d5fc-5e1e-6b13ab2e5321
* Content: [_FLT_PARAMETERS (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/ns-fltkernel-_flt_parameters)
* Content Source: [wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_parameters.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_parameters.md)
* Product: **windows-hardware**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-11-08:
@BrianCatlin , thanks for reporting this. I finally found time to get this done - apologies for the delay!

closed 2019-11-08 by lorihollasch

BrianCatlin, 2019-11-08:
Thanks for this, but there needs to be a description on WHEN and WHY these pseudo IRPs are being sent, and some implementation tips on how to process them. File systems are complex and largely opaque with a very steep - nearly vertical - learning curve, so more information on how to process these IRPs is desperately needed! Without an understanding of when and why these IRPs are sent, driver writers won't understand whether or not to support them and how to support them

lorihollasch, 2019-11-09:
@BrianCatlin , I've made an internal task with a note of your request, although I'm afraid that it won't be actionable in the near future. The team doesn't like to be overly prescriptive given that every filter will behave differently (although I certainly understand that there are always basic rules of the road). Hopefully you are able to glean a good amount of information from the various sample filters that are available. You are certainly welcome to attend the next Plug Fest event that the team sponsors (probably in the Jan/Feb timeline)!