MicrosoftDocs/windows-driver-docs-ddi #408: Not returning information or documented status when device in low-power state

BrianCatlin, 2019-03-01:
I'm getting status = ERROR_FILE_NOT_FOUND when issuing this IOCTL to a port that is in a low-power state. This IOCTL returns registry information, why would this fail if the device is in low-power?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: da5865a6-fe9e-7436-d8f4-fe51a149b2c8
* Version Independent ID: c56fc34b-a7aa-11a0-7fd2-11633ac55936
* Content: [IOCTL_USB_GET_NODE_CONNECTION_DRIVERKEY_NAME (usbioctl.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_driverkey_name)
* Content Source: [wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_driverkey_name.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ni-usbioctl-ioctl_usb_get_node_connection_driverkey_name.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-03-02:
Hi @BrianCatlin, good question. Let me see what I can find out for you to make sure the doc is as accurate as possible.

duncanmacmichael, 2019-03-05:
Following up on emails behind the scenes with the USB team, I'm closing this issue as the documentation is accurate.

closed 2019-03-05 by duncanmacmichael