MicrosoftDocs/windows-driver-docs-ddi #409: USB Constants and Enumerations

Shyamal-V, 2019-03-03:
**USB Constants and Enumerations** link is broken

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: edeb7af8-5aff-1f28-3c35-c4e61649487e
* Version Independent ID: 170aa788-f46d-9a26-8fb3-5b0c413cca8a
* Content: [_USB_CONNECTION_STATUS (usbioctl.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ne-usbioctl-_usb_connection_status)
* Content Source: [wdk-ddi-src/content/usbioctl/ne-usbioctl-_usb_connection_status.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ne-usbioctl-_usb_connection_status.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-03-05:
Hi @Shyamal-V, thanks for finding this. It looks like the page to which that link used to go was actually just a simple portal page that only listed a table of enums for USB drivers in general, and didn't contain any other information. Many of such portal pages were redirected to the top of the WDK reference when we migrated systems last year. Would redirecting this link to the [Enumerations section of the top reference node for USB](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/_usbref/#enumerations) be helpful?

duncanmacmichael, 2019-03-13:
Resolving this due to lack of activity. I updated the link in question to point at the [Enumerations section of the top reference node for USB](https://docs.microsoft.com/windows-hardware/drivers/ddi/content/_usbref/#enumerations), so hopefully that helps.  Those changes should be live later today. Thanks!

closed 2019-03-13 by duncanmacmichael