MicrosoftDocs/windows-driver-docs-ddi #410: What states are valid?

BrianCatlin, 2019-03-04:
You should specify in which states the device can be accessed. It would appear that only DeviceConnected is valid, but it should be stated if the device can be accessed (and why type of access is allowed) in other states

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: edeb7af8-5aff-1f28-3c35-c4e61649487e
* Version Independent ID: 170aa788-f46d-9a26-8fb3-5b0c413cca8a
* Content: [_USB_CONNECTION_STATUS (usbioctl.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/usbioctl/ne-usbioctl-_usb_connection_status)
* Content Source: [wdk-ddi-src/content/usbioctl/ne-usbioctl-_usb_connection_status.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/usbioctl/ne-usbioctl-_usb_connection_status.md)
* Product: **windows-hardware**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

duncanmacmichael, 2019-03-05:
Hi @BrianCatlin, thanks for the question. I also think DeviceConnected would be the only logical choice for accessing the device, but let me see if I can get some confirmation for you.

duncanmacmichael, 2019-03-13:
Hi Brian, still working on getting an answer for you. Thanks!

closed 2019-05-15 by duncanmacmichael