MicrosoftDocs/windows-driver-docs-ddi #412: The docs are poisoned with html escapes

Kojoley, 2019-03-05:
I wanted to open a PR fixing `&amp;` on one of the pages (https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-iocreatefile), but find out there are bunch html escapes all over the place.

EliotSeattle, 2019-03-05:
Thanks, @Kojoley. We became aware of these -- and the extent of the problem -- this week. I am working through some of them, and we're working on solutions for a couple of special, special cases.  I'll leave this open to track this in our reference repo.

tedhudek, 2019-06-11:
Thanks for the sharp eye @Kojoley. I just pushed fixes for ampersand and greater than/less than, across the DDI docs.  Changes should be visible within an hour or two.  Sorry for taking so long!

closed 2019-06-11 by tedhudek