MicrosoftDocs/windows-driver-docs-ddi #415: There is no written priority level for this API

Grabelkovsky, 2019-03-10:
I guess it's PASSIVE.
But it has to be written for every kernel API!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3b739c4a-a8c7-f404-81c3-0b75ff474db2
* Version Independent ID: 1212dcb2-bb17-cc89-f9af-b311d3b828ca
* Content: [AuxKlibQueryModuleInformation function (aux_klib.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/aux_klib/nf-aux_klib-auxklibquerymoduleinformation#feedback)
* Content Source: [wdk-ddi-src/content/aux_klib/nf-aux_klib-auxklibquerymoduleinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/aux_klib/nf-aux_klib-auxklibquerymoduleinformation.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-06-18:
Hi @Grabelkovsky thanks for the feedback and apologies for the delay.

Yes, you are right, this one is passive.  Your best bet for the ones that aren't tagged explicitly on the doc page is to look at the annotations in the header.  Here we have:

```
_IRQL_requires_max_(PASSIVE_LEVEL)
NTSTATUS
__stdcall
AuxKlibQueryModuleInformation (
    _Inout_ PULONG BufferSize,
    _In_ ULONG ElementSize,
    _Out_writes_bytes_opt_(*BufferSize) PVOID QueryInfo
    );
```

I'll make the update to the doc page.

tedhudek, 2019-06-18:
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/aux_klib/nf-aux_klib-auxklibquerymoduleinformation is updated.

closed 2019-06-18 by tedhudek