MicrosoftDocs/windows-driver-docs-ddi #418: Issues with displaying the "Remarks" section.

PawelWMS, 2019-03-12:
Instead of the regular, header-like formatting for the word "Remarks" at the beginning of that section, the string "## Remarks" is being displayed.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d6ca439b-7edb-a39b-c5a5-2e9aabb0beae
* Version Independent ID: e9e00d5a-00ff-ae74-1f08-3dde79ec588a
* Content: [_WDF_DEVICE_POWER_POLICY_IDLE_SETTINGS (wdfdevice.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfdevice/ns-wdfdevice-_wdf_device_power_policy_idle_settings)
* Content Source: [wdk-ddi-src/content/wdfdevice/ns-wdfdevice-_wdf_device_power_policy_idle_settings.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfdevice/ns-wdfdevice-_wdf_device_power_policy_idle_settings.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-05-01:
Thanks for the feedback @PawelWMS and apologies for the delay.  I do not currently repro what you report, so maybe this got fixed somehow?  Please reopen if you still see the problem.

closed 2019-05-01 by tedhudek

PawelWMS, 2019-05-09:
No worries, I also don't see it right now, but I'll be sure to let you know, if this happens again.