MicrosoftDocs/windows-driver-docs-ddi #419: APCs

alfaunits, 2019-03-16:
There is no mention about whether APCs may be disabled, but Driver Verifier complains if they are and FltCreateFileEx is called.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ab2c39da-36d8-c2c3-c978-eae7733de3c1
* Version Independent ID: db182921-736b-9a62-1af3-79d2b5fb392c
* Content: [FltCreateFileEx function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltcreatefileex)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatefileex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltcreatefileex.md)
* Product: **windows-hardware**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

craigbarkhouse, 2019-05-15:
Alfaunits, Driver Verifier only complains if special kernel APCs are disabled (i.e. you are in a guarded region or IRQL >= APC_LEVEL).  It's fine if user APCs and normal kernel APCs are disabled (i.e. you are in a critical region).

lorihollasch, 2019-05-16:
@alfaunits, see @craigbarkhouse 's comment above. We typically don't call out such details on every refpage because this is expected behaviour. See [Disabling APCs](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/disabling-apcs) for more information. Let us know if you have any further questions, thanks!

closed 2019-05-16 by lorihollasch

alfaunits, 2019-05-16:
I think the issue was raised because the debug print for the FltMgr assert is wrong in Windows 7 (it prints out the wrong data for some of the current values), so I presumed that general APCs were at fault
All is fine.