MicrosoftDocs/windows-driver-docs-ddi #425: This page is messed up.

jaykrell, 2019-03-21:
There is an extra underscore on the trailing typedef name, and there is tons of extra content presented. The presentation forgot to stop at the end of the content or something.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e4fa42b5-1632-42e6-8d7d-05778b6eeaa3
* Version Independent ID: ad17836c-e33f-a306-6912-e5ad12d46c6a
* Content: [_FSRTL_ADVANCED_FCB_HEADER (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header#feedback)
* Content Source: [wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/ns-ntifs-_fsrtl_advanced_fcb_header.md)
* Product: **windows-hardware**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2019-03-21:
Thanks, @jaykrell. That is DEFINITELY an issue in our automated build process. 
@lorihollasch can you please investigate with @barrygolden and possibly Dave? I thought we were clear of #if #def errors.
Thanks!

lorihollasch, 2019-05-31:
@jaykrell, thanks again for reporting this. There is a bug in our tools that parse headers to build the docset. We have reported this  issue but have no time estimate for when it will be fixed. Apologies for inconvenience of having to consult the header file directly until this is fixed!

lorihollasch, 2019-06-19:
@jaykrell, since we have an internal bug filed regarding this page, I'm going to close this GitHub issue. Thanks again!

closed 2019-06-19 by lorihollasch

lorihollasch, 2019-07-29:
@jaykrell , FYI that the rendering of this page is now fixed. Thanks again for reporting!