MicrosoftDocs/windows-driver-docs-ddi #427: Address descriptions are optional, if the bus driver doesn't use it

rahul-ramadas, 2019-03-21:
Since the bus driver decides whether it uses address descriptions or not, OldAddressDescription and NewAddressDescription may be null. The documentation should call this out, and the SAL annotations should also probably be corrected.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c49f3fed-c9fb-2fcd-c373-545f08650e25
* Version Independent ID: 516b854e-e900-710c-8540-d3b7709ec286
* Content: [EVT_WDF_CHILD_LIST_DEVICE_REENUMERATED (wdfchildlist.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfchildlist/nc-wdfchildlist-evt_wdf_child_list_device_reenumerated)
* Content Source: [wdk-ddi-src/content/wdfchildlist/nc-wdfchildlist-evt_wdf_child_list_device_reenumerated.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfchildlist/nc-wdfchildlist-evt_wdf_child_list_device_reenumerated.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-09-27:
Hi @rahul-ramadas apologies for the delay.  I made some changes in https://github.com/MicrosoftDocs/windows-driver-docs-ddi/commit/ad2e2f61df06925937be07817050f4c4e00d13b0, can you please let me know if you think it is sufficient?  I left the SAL values as-is because that it what I see in the header, and I think docs should match the header.  (Assume you were thinking of adding _optional_ to the two description parms)  Thanks!

tedhudek, 2019-09-27:
Going to go ahead and close this out but please reopen if you feel more is needed.

tedhudek, 2019-09-27:
BTW updates are now reflected on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfchildlist/nc-wdfchildlist-evt_wdf_child_list_device_reenumerated

closed 2019-09-27 by tedhudek