MicrosoftDocs/windows-driver-docs-ddi #428: ZwSetInformationFile  doesn't mention FileDispositionInformationEx

binarycrusader, 2019-03-22:
The page for FILE_DISPOSITION_INFORMATION_EX now links to ZwSetInformationFile as expected as noted in issue #352 :

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddk/ns-ntddk-_file_disposition_information_ex

However, that page doesn't mention FileDispositionInformationEx as one of the possible FileInformationClass values, and it should as far as I can tell.

metathinker, 2019-03-22:
This is a little interesting - there are actually 2 issues here.

First, the FILE_DISPOSITION_INFORMATION_EX has two links to "ZwSetInformationFile". One actually points to [the page for ZwSetInformationFile](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-zwsetinformationfile), while the other wrongly points to [the page for NtSetInformationFile](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntsetinformationfile).

Second, I didn't realize that we had separate doc pages for ZwSetInformationFile and NtSetInformationFile (and seemingly we do for every Zw/Nt pair, which is really unfortunate IMO), and so @tedhudek and I only added the discussion of FileDispositionInformationEx to the NtSetInformationFile page.

It should be trivial to fix both issues, but it's above my pay grade whether to dedupe the Zw/Nt in general.

tedhudek, 2019-03-22:
Thanks for the internal PR @metathinker.  Results will be live within an hour or so.  Regarding Zw/Nt, [this page](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/using-nt-and-zw-versions-of-the-native-system-services-routines) and children describes the issue more in depth.  The ref docs are built using header reflection, so we're tied for now to the fact that these have separate incarnations in headers.

closed 2019-03-22 by tedhudek