MicrosoftDocs/windows-driver-docs-ddi #429: Compression format COMPRESSION_FORMAT_XPRESS_HUFF issue

UEZUEZ, 2019-03-22:
Somehow the compression format COMPRESSION_FORMAT_XPRESS_HUFF does not work properly either during compression or decompression.

I'm getting always C00000E8: STATUS_INVALID_USER_BUFFER.

Is there a bug with that format?

COMPRESSION_FORMAT_LZNT1 and COMPRESSION_FORMAT_XPRESS are working properly with same code.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5553dc31-7364-ae5d-ff4d-bf1a1757bff8
* Version Independent ID: ca77dfd1-7d40-8240-125a-d02291684cb0
* Content: [RtlCompressBuffer function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-rtlcompressbuffer#feedback)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-rtlcompressbuffer.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-rtlcompressbuffer.md)
* Product: **windows-hardware**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

UEZUEZ, 2019-03-22:
If you want to see the code, here the link: https://www.freebasic.net/forum/viewtopic.php?f=6&p=259395#p259395

UEZUEZ, 2019-03-22:
Ok, it is not a bug, rather my ignorance to read RtlDecompressBuffer function carefully.
RtlDecompressBuffer function does not support the decompression of COMPRESSION_FORMAT_XPRESS_HUFF compressed data - only RtlDecompressBufferEx function does.

closed 2019-03-22 by UEZUEZ

lorihollasch, 2019-03-22:
@U-E-Z , glad you got it figured out! :)