MicrosoftDocs/windows-driver-docs-ddi #431: IRQL is misleading

jtippet, 2019-03-25:
The IRQL docs should be updated to say this:
IRQL: Any IRQL if the BitMapHeader is resident.  Otherwise &lt;= APC_LEVEL if the BitMapHeader is pageable.
Windows 7 and earlier: IRQL must be &lt;= APC_LEVEL

(The discussion about being "resident" is kind of weird, because it's usually an implicit rule of any kernel API: if you give it pageable inputs, the API couldn't possibly run at DISPATCH_LEVEL.  So you can probably even safely omit the whole discussion, since calling it out specifically for this one API is possibly alarming to someone who's reading the docs.)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5d8c4c03-9e6e-d1e2-e3c8-f6cf020f668d
* Version Independent ID: 39bd5fed-6001-0e74-f4ac-085ec5636b13
* Content: [RtlInitializeBitMap function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlinitializebitmap#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-rtlinitializebitmap.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-rtlinitializebitmap.md)
* Product: **windows-hardware**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-06-20:
Hi @jtippet thanks for the suggestion, and apologies for the delay.  I just made ba8340bfc5 to address this.  That commit should sync over to this public mirror within half an hour, then please see what you think.

Also you might know but note for future use you can PR into this point.  To start just click Edit on any page on docs.microsoft.com

tedhudek, 2019-06-28:
Fix is live at https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-rtlinitializebitmap.  Thanks again!

closed 2019-06-28 by tedhudek