MicrosoftDocs/windows-driver-docs-ddi #433: Pre-create normalized name query can return completely unwanted results

alfaunits, 2019-03-31:
If a file name contains a wildcard, which is fairly common when Command Prompt is used (in IRP_MJ_CREATE PreOp), FltMgr will return what would be an invalid result.
A note should be added that wildcards can result in an actual file name being returned, as if a directory listing is performed, and not an IRP_MJ_CREATE.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 82c69953-3606-33d2-1c71-fa25b72a82da
* Version Independent ID: 67b78c7c-0fdc-ab17-d7d7-53a6b3eaded0
* Content: [FltGetFileNameInformation function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltgetfilenameinformation#see-also)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetfilenameinformation.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetfilenameinformation.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-05-18:
@alfaunits , can you submit a code snippet that shows FltMgr returning an invalid result due to a wildcard? It is the application's responsibility to handle the wildcard, resulting in a call to FltGetFileNameInformation for every individual file that the wildcard encapsulates (FltGetFileNameInformation doesn't receive the wildcard). I'm going to close this for now - we can reopen if needed.

closed 2019-05-18 by lorihollasch

alfaunits, 2019-05-18:
Suppose there is a file called Test in C:\Folder.
If any open is attempted for C:\Folder\t*, FltGetFileNameInformation WILL return C:\Folder\Test, instead of STATUS_OBJECT_NAME_INVALID (or similar) code.

This is a side effect from how FltMgr handles normalized file names. I have raised the issue on OSR's file system development forums as well:
https://community.osr.com/discussion/291208/another-fltgetfilenameinformation-bug#latest
and peeps agree. Here's how to repro (from my post on that forum):
[quote]If the CreatePreOp receives a file object whose file name is \?*
(an invalid file name, as passing the I/O through returns
STATUS_OBJECT_NAME_INVALID), on Windows 7, it will still get the file
name that matches that pattern!

This is easily observeable via Command Prompt, just type "cd d" and .
Process Monitor and Zezula's File Spy also exhibit the same file name
output, when ran alone, so it is nothing from our filter.
I traced the IRP_MJ_CREATE and I can see that the file name in the
FltObjects->TargetFileObject is "C:\USers\Dejan\d*", but
FltGetFileNameInformation retrieves C:\Users\Dejan\Desktop as the file
name. [/quote]

alfaunits, 2019-05-18:
There is zero chance this will ever get fixed, as it happens on Windows 7 mostly.
Therefore the documentation should contain a note about a normalized file name query possibly returning a completely unexpected file name should the file object\s file name contain a wildcard, and thus produce unwanted or incorrect behavior.