MicrosoftDocs/windows-driver-docs-ddi #440: KbCallbackRemovePages is not documented

dl2n, 2019-04-05:
KbCallbackRemovePages is not documented yet.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ba2f478f-13c5-44cf-1467-bb5f7513975f
* Version Independent ID: e6195a54-1c53-c616-e8c1-ad6d361eec8a
* Content: [KeRegisterBugCheckReasonCallback function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-keregisterbugcheckreasoncallback#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-keregisterbugcheckreasoncallback.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-keregisterbugcheckreasoncallback.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-05-01:
Thanks @dl2n 

TODO: remove enum vals from https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-keregisterbugcheckreasoncallback and add missing enum vals to https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ne-wdm-_kbugcheck_callback_reason

tedhudek, 2019-06-27:
Hi @dl2n, I've added the description you requested, as well as a bunch of extra info across the docs on using bug check reason callbacks.  Thanks for the feedback, hope this helps!

closed 2019-06-27 by tedhudek