MicrosoftDocs/windows-driver-docs-ddi #441: Error in Parameters

mrolle45, 2019-04-09:
The 'Tag' parameter description should be the same as with ExAllocatePoolWithTag, namely, that the Tag consists of 1 to 4 characters, and the range 0x20 (space) to 0x126 (tilde) applies to each character.  A Tag of 0 is illegal (or at least Driver Verifier doesn't allow it).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d3a6cb22-0679-752f-e6c5-89cb4d3782da
* Version Independent ID: aaa9d239-876d-2ba1-2f13-d91fa60ccbee
* Content: [ExFreePoolWithTag function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-exfreepoolwithtag#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exfreepoolwithtag.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exfreepoolwithtag.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

SIISII, 2019-10-12:
And what about situation when a tag in ExFreePoolWithTag is not equal to tag in ExAllocatePoolWithTag? BSOD on debug-mode kernel and ignore on "normal" kernel?

tedhudek, 2019-10-14:
Thanks @mrolle45 I just pushed an update to make the Tag description for the two pages match.

@SIISII I'm not sure what would happen in that scenario.  Is there a scenario where you are doing this and you think it should be in the docs?

SIISII, 2019-10-14:
> @SIISII I'm not sure what would happen in that scenario. Is there a scenario where you are doing this and you think it should be in the docs?
I think that ideal docs must describe all possible scenarios including such one :) My personal opinion is that using a different tags for allocating and deallocating memory must cause BSOD because this can help to debug drivers and other kernel-mode code. But it seems the kernel ignores these tags at all.

tedhudek, 2019-10-14:
Thanks @SIISII.  Agreed more info is better.  Since it seems the result of a tag mismatch is a no-op, I'm going to close this one out.  Thanks again for caring about the docs :)

closed 2019-10-14 by tedhudek