MicrosoftDocs/windows-driver-docs-ddi #442: Tag cannot be 0

mrolle45, 2019-04-09:
The Parameters section on the Tag parameter should specify "1 to 4" characters, rather than "up to 4" characters.
0 characters cannot be represented as a character literal, but an integer value of 0 could be supplied.  I don't know what the effect would be, but Driver Verifier flags this as an error.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: edc6540c-b149-ed6a-49b3-25ad080fb719
* Version Independent ID: d6cb600f-2879-8455-43db-d80e664e24e6
* Content: [ExAllocatePoolWithTag function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-exallocatepoolwithtag)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-exallocatepoolwithtag.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-exallocatepoolwithtag.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

SMBB, 2019-05-26:
Up until Windows version 1909 you were able to pass 0 as a valid tag (if you ignore the Driver Verifier error). However, since the release of 1909 it will just refuse to allocate the pool all together.

tedhudek, 2019-08-22:
See #567.  Thanks for the reports and apologies for the delay.

closed 2019-08-22 by tedhudek