MicrosoftDocs/windows-driver-docs-ddi #446: UsageList cannot be null

lgonzalezluna, 2019-04-12:
The documentation currently says "Alternatively, applications or drivers can call HidP_GetUsages and set UsageList to zero to return the required length in UsageLength" but doing so just causes the application to crash (Windows 10).

Calling  HidP_MaxUsageListLength() works though.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 535caf6e-4209-084e-e110-3f94ec42f252
* Version Independent ID: 7ff2a064-c75e-bca4-8abe-3d01010b869c
* Content: [HidP_GetUsages function (hidpi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/hidpi/nf-hidpi-hidp_getusages#feedback)
* Content Source: [wdk-ddi-src/content/hidpi/nf-hidpi-hidp_getusages.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/hidpi/nf-hidpi-hidp_getusages.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

duncanmacmichael, 2019-04-16:
Hi @lgonzalezluna, thanks for pointing this out. Let me see if I can reach out to the devs and confirm this for you.

duncanmacmichael, 2019-04-16:
Hi @lgonzalezluna, I've gotten a response from the devs on this. It does look like we have a typo when referring to the correct parameter here. To get the required length, (*UsageLength) should be zero. In other words, UsageLength should be a valid pointer that points to a ULONG value **0**. I'll go ahead and update the doc, and those changes should be live later today. Thanks!

closed 2019-04-16 by duncanmacmichael