MicrosoftDocs/windows-driver-docs-ddi #447: This page no longer works in IE11

alfaunits, 2019-04-13:
The left pane (navigation) overlaps virtually all of the right side.

This happens on ALL of drivers/ddi/content as of a few days ago!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 020480f3-b724-f282-cce8-43d7cde038d2
* Version Independent ID: 84eeb4e3-4d88-862c-276e-7496c0f35f9b
* Content: [_FLT_REGISTRATION (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/ns-fltkernel-_flt_registration)
* Content Source: [wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_registration.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/ns-fltkernel-_flt_registration.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-04-15:
@alfaunits,  thanks for reporting. We are unable to repro this issue - have tried using several versions of IE11 on different machines. Am closing this for now - let me know if you are still seeing it.

closed 2019-04-15 by lorihollasch

alfaunits, 2019-04-15:
You meed Large Fonts, or smaller resolution. I use that as I am visually
impaired, and IE is the only browser which doesnt hurt :)

alfaunits, 2019-04-16:
Heh, I just rechecked... now it is back to normal, with absolutely no scaling/font changes since this morning. Cheers!