MicrosoftDocs/windows-driver-docs-ddi #449: Bad syntax example

BrianCatlin, 2019-04-15:
The definition of PFLT_FILTER_UNLOAD_CALLBACK is a pointer to a function - not a function prototype, as is implied by the usage in the Syntax section. As far as I can tell, there isn't a function prototype defined for the unload routine, which is lazy and indicates poor design (contrast this with WDF, which does define function prototypes for all callback routines, and it even include SAL annotations!).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f522a783-c1d1-b3b0-6845-f3cb391ae3ca
* Version Independent ID: f855f235-4e1c-3506-0c4e-3fe0559ad842
* Content: [PFLT_FILTER_UNLOAD_CALLBACK (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nc-fltkernel-pflt_filter_unload_callback)
* Content Source: [wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_filter_unload_callback.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nc-fltkernel-pflt_filter_unload_callback.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

BrianCatlin, 2019-04-16:
This problem also exists for all the other routines whose pointers are in FLT_REGISTRATION (e.g. PFLT_INSTANCE_SETUP_CALLBACK) and are linked from the FLT_REGISTRATION page

1-0-9-1, 2019-04-16:
This

BrianCatlin, 2019-04-17:
1-0-9-1, this is for file system minifilters; it has nothing to do with networking

craigbarkhouse, 2019-05-01:
Are you recommending that it be defined like this in the header file:

typedef NTSTATUS
(__stdcall FLT_FILTER_UNLOAD_CALLBACK) (
    FLT_FILTER_UNLOAD_FLAGS Flags
    );
typedef FLT_FILTER_UNLOAD_CALLBACK *PFLT_FILTER_UNLOAD_CALLBACK;

As opposed to how it is defined now:

typedef NTSTATUS
(__stdcall *PFLT_FILTER_UNLOAD_CALLBACK) (
    FLT_FILTER_UNLOAD_FLAGS Flags
    );

And if so, what is the practical benefit?

BrianCatlin, 2019-05-01:
What possible use would a driver writer have for a pointer of this type? Instead, it should be defined as follows, for use as a forward declaration, which a driver writer definitely needs.

typedef
_Function_class_ (FLT_FILTER_UNLOAD_CALLBACK)
_IRQL_requires_same_
_IRQL_requires_max_(PASSIVE_LEVEL)
NTSTATUS
FLT_FILTER_UNLOAD_CALLBACK
                (
                _In_
                FLT_FILTER_UNLOAD_FLAGS      Flags
                );

A pointer could then be derived from the prototype, like this:

typedef FLT_FILTER_UNLOAD_CALLBACK *PFN_FLT_FILTER_UNLOAD_CALLBACK;

This is the way that WDF does things, and the other driver models should follow because it provides the most functionality.

lorihollasch, 2019-05-10:
@BrianCatlin , thanks for submitting this issue. I am closing it since the documentation itself is correct. Your suggestion has been received by the FS dev team via @craigbarkhouse , and they will decide whether they are able to make your suggested changes.

closed 2019-05-10 by lorihollasch