MicrosoftDocs/windows-driver-docs-ddi #455: EX_CALLBACK_FUNCTION callback function: RegNtPostRestoreKey wrong structure type

djeauh, 2019-04-18:
The documentation says that RegNtPostRestoreKey notification class should be handled using a REG_RESTORE_KEY_INFORMATION. We experienced BSOD while trying to use this structure because fields were not valid:

1: kd&gt; dt _REG_RESTORE_KEY_INFORMATION poi(Argument2)
_REG_RESTORE_KEY_INFORMATION
+0x000 Object : 0xffffb205ed1c9750 Void
+0x008 FileHandle : 0x00000000c0000022 Void ----&gt; This is an invalid handle. It seems like an error NTSTATUS
+0x010 Flags : 0xb9c53a50 ----&gt; Flags are invalid
+0x018 CallContext : 0x00000000c0000022 Void ----&gt; Should be NULL in our case since we did not set any call context.
+0x020 ObjectContext : (null)
+0x028 Reserved : 0xffffb205ee496550 Void

Mapping the argument to a _REG_RESTORE_KEY_INFORMATION seems better:

1: kd&gt; dt _REG_POST_OPERATION_INFORMATION poi(Argument2)
+0x000 Object : 0xffffb205ed1c9750 Void
+0x008 Status : 0n-1073741790 -----------&gt; 0xc0000022
+0x010 PreInformation : 0xffffef07b9c53a50 Void
+0x018 ReturnStatus : 0n-1073741790 -----------&gt; 0xc0000022
+0x020 CallContext : (null)
+0x028 ObjectContext : 0xffffb205`ee496550 Void
+0x030 Reserved : (null)

1: kd&gt; dt _REG_RESTORE_KEY_INFORMATION 0xffffef07`b9c53a50
+0x000 Object : 0xffffb205ed1c9750 Void
+0x008 FileHandle : 0xffffffff8009e714 Void
+0x010 Flags : 8
+0x018 CallContext : (null)
+0x020 ObjectContext : (null)
+0x028 Reserved : (null)

Could you confirm the error please?

What about the other structures like REG_SAVE_KEY_INFORMATION/REG_REPLACE_KEY_INFORMATION?

Thank you for your help!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bb5182bb-69f3-8586-1b4e-270701fb4451
* Version Independent ID: d695c472-d297-4931-6841-f60c13d83c57
* Content: [EX_CALLBACK_FUNCTION (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nc-wdm-ex_callback_function)
* Content Source: [wdk-ddi-src/content/wdm/nc-wdm-ex_callback_function.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nc-wdm-ex_callback_function.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

djeauh, 2019-04-24:
I confirm that the same documentation issue exists for the RegNtPostSaveKey and RegNtPostReplaceKey operation classes.

To summarize the issue, REG_RESTORE_KEY_INFORMATION structure MUST be used for RegNtPostRestoreKey/RegNtPostSaveKey/RegNtPostReplaceKey operation classes, instead of the structures mentionned in the documentation (REG_RESTORE_KEY_INFORMATION/REG_SAVE_KEY_INFORMATION/REG_REPLACE_KEY_INFORMATION)

tedhudek, 2019-06-28:
Hi @djeauh thanks so much for the feedback and apologies for the delay.

I'm not clear what you are suggesting for RegNtPostRestoreKey.  This value already has REG_RESTORE_KEY_INFORMATION struct listed for it on https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nc-wdm-ex_callback_function.

So to summarize, you are suggesting:

|value|struct|
|-|-|
|RegNtPostRestoreKey|REG_RESTORE_KEY_INFORMATION (already listed)|
|RegNtPostSaveKey|REG_RESTORE_KEY_INFORMATION (change from current REG_SAVE_KEY_INFORMATION)|
|RegNtPostReplaceKey|REG_RESTORE_KEY_INFORMATION (change from current REG_REPLACE_KEY_INFORMATION)|

Is that right?

djeauh, 2019-07-01:
Hello @tedhudek,

    I'm a dumb... I made a mistake writing my posts! The structure REG_POST_OPERATION_INFORMATION should be used for RegNtPostRestoreKey/RegNtPostSaveKey/RegNtPostReplaceKey instead of REG_RESTORE_KEY_INFORMATION/REG_SAVE_KEY_INFORMATION/REG_REPLACE_KEY_INFORMATION.

No problem for the late answer. Let me know if it is clear for you.

Note : I'm going to fix the posts to remove ambiguities...

tedhudek, 2019-07-19:
Thanks for the clarification, @djeauh I've made the change.  Give it an hour or two to propagate to the live site.

closed 2019-07-19 by tedhudek