MicrosoftDocs/windows-driver-docs-ddi #456: BaseAddress  vs. *BaseAddress

mikrosk, 2019-04-23:
Shouldn't this sentence:

"If the MEM_RELEASE flag is set in the FreeType parameter, BaseAddress must be the base address returned by ZwAllocateVirtualMemory when the region was reserved."

be:

"If the MEM_RELEASE flag is set in the FreeType parameter, *BaseAddress must be the base address returned by ZwAllocateVirtualMemory when the region was reserved."

?

There is a couple of similar issues in the text, same as for NtAllocateVirtualMemory() description -- the text should make it clear that it is always *BaseAddress and *RegionSize you are talking about, not the pointers themselves.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dc08af12-0480-fba3-7498-98545e6a62a4
* Version Independent ID: f5771ed8-3447-68a9-e548-0709cfd13587
* Content: [NtFreeVirtualMemory function (ntifs.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntfreevirtualmemory)
* Content Source: [wdk-ddi-src/content/ntifs/nf-ntifs-ntfreevirtualmemory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/ntifs/nf-ntifs-ntfreevirtualmemory.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-04-27:
@mikrosk , thanks for flagging this. The language was indeed unclear - fixed now in in NtAllocateVirtualMemory and NtFreeVirtualMemory.

mikrosk, 2019-04-27:
@lorihollasch thanks for such prompt response! I don't want to ruin the moment but it is still not perfect. ;)

In NtFreeVirtualMemory:

_If the MEM_DECOMMIT flag is set in the FreeType parameter, NtFreeVirtualMemory decommits all memory pages that contain one or more bytes in the range from the BaseAddress parameter to (BaseAddress + *RegionSize)_

[...]

_If you specify this flag, *RegionSize must be zero, and BaseAddress must point to the base address returned by NtAllocateVirtualMemory when the region was reserved. NtFreeVirtualMemory fails if either of these conditions is not met._

Also, while we are on the topic:

_You can use NtAllocateVirtualMemory to put committed memory pages into either the reserved or free state._

"NtAllocateVirtualMemory" in this sentence (and few others) points for some reason here: https://msdn.microsoft.com/library/windows/hardware/ff566416 (`ZwAllocateVirtualMemory`) while it should point here: https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntallocatevirtualmemory (`NtAllocateVirtualMemory`).

Also it is quite strange that https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntifs/nf-ntifs-ntallocatevirtualmemory (`NtAllocateVirtualMemory`) refers to itself as `ZwAllocateVirtualMemory`, while `NtFreeVirtualMemory` is consistent in this matter.

lorihollasch, 2019-06-11:
@mikrosk , so sorry! Just now seeing that you provided additional feedback after I closed this. I'll reopen and look at this within the next couple of weeks. :)

mikrosk, 2019-06-11:
No problem Lori, if you need something to clarify more, feel free to ask. :)

lorihollasch, 2019-06-27:
@mikrosk, think I got all the pointer language this time, and also fixed copy-and-paste Zw errors - thanks for pointing them out!

closed 2019-06-27 by lorihollasch