MicrosoftDocs/windows-driver-docs-ddi #458: Formatting and indentation for code snippet is broken

Shyamal-V, 2019-04-24:
Formatting and indentation for code snippet is broken

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6f9f298b-0cdc-d05b-dcdd-5005071b8e23
* Version Independent ID: cec2fc09-d2da-590e-3972-88e055e9d61d
* Content: [WdfIoTargetWdmGetTargetFileHandle function (wdfiotarget.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdfiotarget/nf-wdfiotarget-wdfiotargetwdmgettargetfilehandle)
* Content Source: [wdk-ddi-src/content/wdfiotarget/nf-wdfiotarget-wdfiotargetwdmgettargetfilehandle.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdfiotarget/nf-wdfiotarget-wdfiotargetwdmgettargetfilehandle.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-05-01:
@billlattimer try this one. Recommend pulling the code snippets out of HTML tables and using the markdown code fencing, for example:

```cpp
code goes here
```

tedhudek, 2019-10-08:
Fixed

closed 2019-10-08 by tedhudek