MicrosoftDocs/windows-driver-docs-ddi #459: Bad formating for a few items

alfaunits, 2019-04-24:
FileCaseSensitiveInformation
FileStatLxInformation
FileStatInformation
FileDesiredStorageClassInformation
FileStorageReserveIdInformation

Their description is seen as if some bad character god inserted somewhere and messed up the formatting. E.g. :
A [**FILE_CASE_SENSITIVE_INFORMATION**](../ntifs/ns-ntifs-_file_case_sensitive_information.md), declared in ntifs.h. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b2d450a1-f049-5a37-fe23-76b369d89738
* Version Independent ID: 2b528307-4778-6bd7-a10c-d6d518f7df98
* Content: [_FILE_INFORMATION_CLASS (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ne-wdm-_file_information_class)
* Content Source: [wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ne-wdm-_file_information_class.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-05-01:
@billlattimer try this one. The key is that you can't have markdown-style links in HTML. So either switch the links to HTML style, or convert the table to md-style

billlattimer, 2019-06-05:
@alfaunits Dejan, thank you for catching this - corrected the formatting errors and all links

closed 2019-06-05 by billlattimer