MicrosoftDocs/windows-driver-docs-ddi #465: Link to PORT_CONFIGURATION_INFORMATION points to the WRONG structure

arlied-google, 2019-04-26:
I just spent about 2 hours debugging an issue in assembly code, and nothing made sense, until I realized that the docs on this page were wrong.  This page has links with text "PORT_CONFIGURATION_INFORMATION", but they point to the wrong destination. They point to a structure with that named defined in Windows 2000 Kernel Streaming, NOT the version of this structure defined in storport.h.

Please fix this.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: fc813b89-0c63-61ff-3142-5e0297d46814
* Version Independent ID: ca57da37-689e-6997-324f-eab7261938f2
* Content: [StorPortGetDeviceBase function (storport.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/storport/nf-storport-storportgetdevicebase#feedback)
* Content Source: [wdk-ddi-src/content/storport/nf-storport-storportgetdevicebase.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/storport/nf-storport-storportgetdevicebase.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2019-04-26:
Hi @arlied-google.

I'm so sorry that you had issues as a result of the misdirected links, and I thank you for bringing it to my attention. I've made the corrections and submitted a PR for the changes. This should be live later today.

Eliot

closed 2019-04-26 by EliotSeattle

arlied-google, 2019-04-26:
Thanks!  It's appreciated.