MicrosoftDocs/windows-driver-docs-ddi #473: Table in [Return Value] section isn't formatted correctly

kyanha, 2019-05-03:
The table in the [Return Value] section isn't correctly formatted.  See the "Content Source" link to see Github's rendering of the Markdown for that table, which is not correctly rendered and also matches how it is rendered on docs.microsoft.com.

(thank you for spending so much time and effort making these docs as excellent and precise as they are!)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a37359e8-8ea8-7240-7302-e3a1a6225060
* Version Independent ID: 0822ae8c-973d-5051-ea91-e98bc4a0ee12
* Content: [IoGetDeviceDirectory function (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/nf-wdm-iogetdevicedirectory#feedback)
* Content Source: [wdk-ddi-src/content/wdm/nf-wdm-iogetdevicedirectory.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/nf-wdm-iogetdevicedirectory.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-05-03:
Thanks @kyanha for the sharp eye.  Assigning over to Bill for the fix

billlattimer, 2019-10-31:
@kyanha thank you for finding this, and for your kind comments. This is resolved.

closed 2019-10-31 by billlattimer