MicrosoftDocs/windows-driver-docs-ddi #476: Broken links

harvv, 2019-05-08:
The links below this paragraph: "The input report is returned in the ReportBuffer parameter. Depending on the report ID, the caller parses the report by calling one of the following functions:"

...are broken and lead to "404 pages".

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ac5b1697-381b-c035-e1ad-d72a722d8cce
* Version Independent ID: ca8c2411-035f-6f1e-ca27-4526d29eed95
* Content: [HidD_GetInputReport function (hidsdi.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/hidsdi/nf-hidsdi-hidd_getinputreport#feedback)
* Content Source: [wdk-ddi-src/content/hidsdi/nf-hidsdi-hidd_getinputreport.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/hidsdi/nf-hidsdi-hidd_getinputreport.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @duncanmacmichael
* Microsoft Alias: **dumacmic**

EliotSeattle, 2019-05-08:
@duncanmacmichael 
IMHO, The entire header is not represented in the wdk-DDI source. Is this something that ships in the SDK and that we have traditionally linked to? If so, maybe work with Drew to get the header installed on their side?

EliotSeattle, 2019-05-08:
My mistake, @duncanmacmichael. hidpi.h is in the km folder. We're missing it. Can you work with Barry/Dave to rectify? 
Meanwhile, @harvv, while Duncan is working to return our docs to working shape, are there any of those for which you need to see the content quickly? If you get me your email, I can send you the content for any of those pages. And **THANK YOU** for reporting this!

harvv, 2019-05-10:
no need, I am somewhat familiar with the sdk and ddk files so I found them and the information. I wanted to make sure you were aware of the break though
(but thanks for asking!)

duncanmacmichael, 2019-06-08:
Sorry for the delay on this; I actually had another issue ([1585](https://github.com/MicrosoftDocs/windows-driver-docs/issues/1585)) opened up with this same problem, so I responded to that one and forgot to say anything here. I have filed an internal bug to get this fixed, and plan on updating issue 1585 when that is complete. In the meantime I'm going to close this one as it's a duplicate. Thanks!

closed 2019-06-08 by duncanmacmichael