MicrosoftDocs/windows-driver-docs-ddi #482: Namespace and include requirements

billti, 2019-05-10:
This page doesn't mention, but these functions are not available unless you include the appropriate WRL headers first (there is a guard around them `#ifdef _WRL_CLIENT_H_`).

Also worth mentioning is that these are under a namespace (`Debugger::DataModel`).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c59c0773-0f69-dbc4-4754-4ed647160b9c
* Version Independent ID: e02b5af6-9811-3d56-1025-0083e61cde53
* Content: [BindProperty function (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-bindproperty#feedback)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-bindproperty.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-bindproperty.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-11:
@billti - Thanks for this issue. I added additional information to this topic to mention the C++ sample code on git hub that provides this information. I will close this issue when those updates are live.

DOMARS, 2019-06-12:
@billti - The topic updates are live. Thanks for this issue and pointing out what you ran into with the DbgModel.

closed 2019-06-12 by DOMARS