MicrosoftDocs/windows-driver-docs-ddi #484: Code samples use external types

billti, 2019-05-11:
These code samples are a little confusing, as neither TypedInstanceModel or Object are defined in DbgModel.h file (or any files in the SDK), but in the external DbgModelClientEx.h header from GitHub. It should be clear this is required, or the sample changed to only use APIs from the SDK headers.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c59c0773-0f69-dbc4-4754-4ed647160b9c
* Version Independent ID: e02b5af6-9811-3d56-1025-0083e61cde53
* Content: [BindProperty function (dbgmodel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/dbgmodel/nf-dbgmodel-bindproperty)
* Content Source: [wdk-ddi-src/content/dbgmodel/nf-dbgmodel-bindproperty.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/dbgmodel/nf-dbgmodel-bindproperty.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2019-06-11:
@billti - Thanks for this issue. I added additional information to this topic to calll out the use of the as recommended wrapper lib. Will close this issue when those updates are live.

DOMARS, 2019-06-12:
@billti - The updates to the topic are live. Thanks for calling out what you ran into so we can better clarify the use of the recommended wrapper lib.

closed 2019-06-12 by DOMARS