MicrosoftDocs/windows-driver-docs-ddi #485: Need output string details

BrianCatlin, 2019-05-13:
Is the output volume name string zero-terminated? If so, does the Length field in the descriptor include the terminator?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 55e05b21-c927-e661-6ad4-e430eb5d59ac
* Version Independent ID: 3b6c2052-2fa5-5542-5f56-9ab43b3b7846
* Content: [FltGetVolumeName function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltgetvolumename)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetvolumename.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltgetvolumename.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-05-23:
@briancatlin, the answer to this question is on the UNICODE_STRING page, which has disappeared from our doc set and resulted in broken links like the one on this page, sorry! We will work to find the disappeared page and get it back online. In the interim, I'll answer your questions and close this issue. The output volume name string is not null-terminated and length is the length of the non-null-terminated string. Until we get the UNICODE_STRING page up, you can find the info on [NTSCSI_UNICODE_STRING](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/ntddscsi/ns-ntddscsi-ntscsi_unicode_string). Thanks for making us aware of this!

closed 2019-05-23 by lorihollasch

tedhudek, 2019-05-29:
The good news is that the page for [**UNICODE_STRING**](https://docs.microsoft.com/en-us/windows/desktop/api/ntdef/ns-ntdef-_unicode_string) is restored.  The bad news is that there's a known rendering bug with the ref doc pipeline that is messing up the description of the **Buffer** member.