MicrosoftDocs/windows-driver-docs-ddi #486: When and why should this routine be called?

BrianCatlin, 2019-05-13:
Do I need to call this only if my driver is looking at the file data? Are there other reasons? If this routine returns false, what is the recommended way of handling conflicts?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9c70db50-774f-57cb-b250-e6c009492782
* Version Independent ID: c1cdb098-b233-3cf0-d154-f2383e098aa7
* Content: [FltRegisterForDataScan function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltregisterfordatascan)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltregisterfordatascan.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltregisterfordatascan.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-05-31:
@briancatlin, thanks for submitting this - I have updated FltRegisterForDataScan to answer your questions!

closed 2019-05-31 by lorihollasch

BrianCatlin, 2019-05-31:
What is the recommend mechanism for "However, section access is not synchronized and conflict resolution is left to the minifilter driver through its PFLT_SECTION_CONFLICT_NOTIFICATION_CALLBACK callback routine"? Is there an example, somewhere? Also, what are the reasons that it will fail? Does it have to do with specific volume attributes? File system type?