MicrosoftDocs/windows-driver-docs-ddi #492: Missing link

BrianCatlin, 2019-05-19:
Why isn't there a link to the definition of the FileAttributes field? Why must we perform another search for GetFileAttributes?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 30d6a301-2028-bba5-0e43-092824810891
* Version Independent ID: 5c8c373a-b34a-5e73-8002-57076540e6ce
* Content: [_FILE_BASIC_INFORMATION (wdm.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/wdm/ns-wdm-_file_basic_information)
* Content Source: [wdk-ddi-src/content/wdm/ns-wdm-_file_basic_information.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/wdm/ns-wdm-_file_basic_information.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2019-05-20:
Good point.  Feel free to create a PR for such items.

tedhudek, 2019-05-20:
Should be live in ~10 min

closed 2019-05-20 by tedhudek