MicrosoftDocs/windows-driver-docs-ddi #494: Which file systems support contexts?

BrianCatlin, 2019-05-19:
Please provide a table that lists which file systems support contexts and under what conditions (e.g. NTFS doesn't support for page file)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7e305035-0748-94fe-854c-cebed4864b11
* Version Independent ID: f50e4401-cd20-f69f-2865-423948aa4c7c
* Content: [FltSupportsFileContextsEx function (fltkernel.h) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/fltkernel/nf-fltkernel-fltsupportsfilecontextsex)
* Content Source: [wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsupportsfilecontextsex.md](https://github.com/MicrosoftDocs/windows-driver-docs-ddi/blob/staging/wdk-ddi-src/content/fltkernel/nf-fltkernel-fltsupportsfilecontextsex.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2019-05-20:
@BrianCatlin , are you looking for more information beyond the details in the Conceptual material? https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/file-system-support-for-contexts

BrianCatlin, 2019-05-20:
There are 30 file systems in the FLT_FILESYSTEM_TYPE enum. We need a table that describes for every file system the conditions on if/when contexts are supported. We cannot afford to guess and then get a crash because some condition (e.g. pagefile) doesn't support a context.

lorihollasch, 2019-05-30:
@briancatlin, it's not feasible to come up with a comprehensive list of all conditions for all file systems (some of which we have no visibility into). Minifilters need to error-check and not attach a context if a context type isn't supported. I added some language to make that more clear, along with a link to the conceptual information.

closed 2019-05-30 by lorihollasch

BrianCatlin, 2019-05-30:
OK, you don't have visibility into all file systems, but you certainly have visibility into Microsoft-authored file systems. What is the rationale for not documenting those? Trial and error is not a reasonable methodology for writing software!

lorihollasch, 2019-05-31:
Sorry, but a comprehensive table could be more confusing for minifilter developers. Rather than trying to understand and accommodate a list of special cases (and not being able to anticipate and accommodate all special cases for third party systems), it is simpler for minifilters to call this function (or check the status code returned by FltSetInstanceContext as some of the sample drivers do), and not attach if a context type isn't supported.

BrianCatlin, 2019-05-31:
I disagree. If I am designing a minifilter, and I see that file systems that I don't care about, such as CSV or FAT, don't support contexts I can design my minifilter to use contexts, but if I don't know which file systems support contexts, then I have to design a much more complicated mechanism to support all possible file systems with workarounds that will never be used. The docs need to be more explicit because unlike the rest of the driver models, there is too much information about file systems and minifilters that can only be described as "lore", which sets the bar for entry too high for beginners.

craigbarkhouse, 2019-06-05:
Brian, FltSupportsFileContextsEx tells you if a file system supports file contexts in addition to stream contexts.  You can assume that any file system that supports named streams (as per FltQueryVolumeInformation(…,FileFsAttributeInformation,...)) also supports file contexts, because information about a file is distinct from information about a stream.  For file systems that don't support named streams, they still support stream contexts and that's where you can put information about a file given that streams and files are 1:1.  So if your only requirement is you need to keep track of some stuff per file, the lack of FileContexts per se should not be a problem.
The reason FltSupportsFileContextsEx takes a file object is to give file systems the ability to vary this by file.  The paging file example is the only current example I know of based on the inbox file systems, but the ability is there, so file systems could use it.  For example, it's conceivable to have a file system that supports named streams but not on directories, therefore not support file contexts on directories.  That's not true of NTFS (NTFS supports named streams even on directories) but something like that is conceivable.
We don't want to draw up a reference table because besides being a lot of work, we don't want to imply that this is a static thing.  It is dynamic and can change over time.  When coding your filter you should always be thinking in terms of being as flexible as possible and not make assumptions, especially not based on the name of the file system.  I know that's not entirely possible in all cases, simply because there are far too many behavioral differences to ever capture in attribute flags.
Also make use of community knowledge, such as OSR's NTFSD mailing list.  There are plenty of experienced developers out that who have thought about these same problems and are willing to share their solutions.  These doc pages are merely a reference, not a comprehensive tutorial.

BrianCatlin, 2019-06-05:
Thank you for the explanation. Why isn't that documented?
The problem I have with the file system/minifilter docs, is that too much of the necessary information is "lore", and is not contained in Microsoft's documentation - which should be the primary reference. For every other driver model, they document everything and provide a tutorial. Networking manages to provide the necessary documentation and that is far more complex than minifilters. So, why aren't minifilters documented as well? Regarding the OSR site, because that is not run by Microsoft, there is no way to know if the information is correct and there is no recourse if it isn't. Things change over time, and the posts in their forum aren't updated. Also, depending upon third parties to answer questions adds intolerable delays into the development of a driver. The end result is that unlike the other driver models, where I can read the docs and design my driver, in file systems/minifilters I have to use trial and error to develop my driver. Given the importance of these drivers and the confidence that we must have in them to not distort data under any circumstances, is this really the best way to develop them? Why can't the file system/minifilter drivers be documented as well as the network drivers?